moveit/moveit2_tutorials #967: colcon build --mixin release: moveit_task_constructor_core error

2vin2vin, 2024-09-17:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* python version: 3.10
* https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html
Followed the above tutorial with humble as the branch

During colcon build got the below error
Starting >>> moveit
--- stderr: moveit_task_constructor_core
CMake Error at python/pybind11/tools/FindPythonLibsNew.cmake:163 (message):
  Python config failure:

  Traceback (most recent call last):

    File "<string>", line 6, in <module>

  ImportError: cannot import name 'sysconfig'

Call Stack (most recent call first):
  python/pybind11/tools/pybind11Tools.cmake:50 (find_package)
  python/pybind11/tools/pybind11Common.cmake:192 (include)
  python/pybind11/CMakeLists.txt:254 (include)

---
Failed   <<< moveit_task_constructor_core [1.30s, exited with code 1]

I seem to have sysconfig and it seems to work as well

I checked for the file FindPythonLibsNew.cmake and got the following
/home/2vin/.local/lib/python2.7/site-packages/pybind11/share/cmake/pybind11/FindPythonLibsNew.cmake
/home/2vin/.local/lib/python3.8/site-packages/pybind11/share/cmake/pybind11/FindPythonLibsNew.cmake

but I also seem to have the file in 
/home/2vin/.local/lib/python3.10/site-packages/pybind11/share/cmake/pybind11/FindPythonLibsNew.cmake
but not detected in locate

so I have been trying everything I can since 2 days but unable to resolve it.

rhaschke, 2024-09-17:
Looks like you screwed your python environment. I guess, you have multiple python binaries and the wrong one is used.
The cmake file should come from `moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake`.
Maybe you can use `colcon build --cmake-args --trace` to get verbose cmake output and figure out which python binary is used.
As a last resort, I suggest cleaning up `~/.local/lib/python*`. However, this might break other apps!

2vin2vin, 2024-09-17:
/home/2vin/ws_moveit/src/moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake(129):  execute_process(COMMAND ${PYTHON_EXECUTABLE} -c 
import sys;import struct;
import sysconfig as s
USE_SYSCONFIG = sys.version_info >= (3, 10)
if not USE_SYSCONFIG:
    from distutils import sysconfig as ds
print('.'.join(str(v) for v in sys.version_info));
print(sys.prefix);
if USE_SYSCONFIG:
    scheme = s.get_default_scheme()
    if scheme == 'posix_local':
        # Debian's default scheme installs to /usr/local/ but we want to find headers in /usr/
        scheme = 'posix_prefix'
    print(s.get_path('platinclude', scheme))
    print(s.get_path('platlib'))
    print(s.get_config_var('EXT_SUFFIX') or s.get_config_var('SO'))
else:
    print(ds.get_python_inc(plat_specific=True));
    print(ds.get_python_lib(plat_specific=True));
    print(ds.get_config_var('EXT_SUFFIX') or ds.get_config_var('SO'));
print(hasattr(sys, 'gettotalrefcount')+0);
print(struct.calcsize('@P'));
print(s.get_config_var('LDVERSION') or s.get_config_var('VERSION'));
print(s.get_config_var('LIBDIR') or '');
print(s.get_config_var('MULTIARCH') or '');
 RESULT_VARIABLE _PYTHON_SUCCESS OUTPUT_VARIABLE _PYTHON_VALUES ERROR_VARIABLE _PYTHON_ERROR_VALUE )
/home/ubuntu/ws_moveit/src/moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake(161):  if(NOT _PYTHON_SUCCESS MATCHES 0 )
/home/ubuntu/ws_moveit/src/moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake(162):  if(PythonLibsNew_FIND_REQUIRED )
/home/ubuntu/ws_moveit/src/moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake(163):  message(FATAL_ERROR Python config failure:\n${_PYTHON_ERROR_VALUE} )
CMake Error at python/pybind11/tools/FindPythonLibsNew.cmake:163 (message):
  Python config failure:

  Traceback (most recent call last):

    File "<string>", line 6, in <module>

  ImportError: cannot import name 'sysconfig'

Call Stack (most recent call first):
  python/pybind11/tools/pybind11Tools.cmake:50 (find_package)
  python/pybind11/tools/pybind11Common.cmake:192 (include)
  python/pybind11/CMakeLists.txt:254 (include)

---
Failed   <<< moveit_task_constructor_core [22.5s, exited with code 1]
I have also tried to check the python version from python terminal it seems to be 3.10 and also as you said it is being taken from moveit_task_constructor
I have actually upgraded fromubuntu 20 to 22 and then tries to install this and I have been facing issues continuously I have tried to resolve most of them but I am unable to solve this

2vin2vin, 2024-09-17:
> /home/2vin/ws_moveit/src/moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake(129): execute_process(COMMAND ${PYTHON_EXECUTABLE} -c import sys;import struct; import sysconfig [...]

Also distutils.setuptools exists but it seems to be generating deprecated warning

rhaschke, 2024-09-17:
You still didn't figure out which python executable is called by cmake. Try to print the cmake variable `PYTHON_EXECUTABLE` in /home/2vin/ws_moveit/src/moveit_task_constructor/core/python/pybind11/tools/FindPythonLibsNew.cmake:129

2vin2vin, 2024-09-18:
It seems like it uses python3.6
I have tried to set -DPython _EXECUTABLE path/to/python for which i get
manually entered variabled were not used by project Python_EXECUTABLE