moveit/moveit2_tutorials #970: MoveItPy RuntimeError: Planning plugin name is empty or not defined in namespace 'ompl'

elsa9334, 2024-09-18:
### Description

Want to run MoveItPy using a UR arm. 

### Your environment
* ROS Distro: Jazzy
* OS Version: e.g. Ubuntu 22.04
* Source build, main branch

### Steps to reproduce
First the robot is connected to the computer using this command. The launch file was downloaded from [UR Robot](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main) and no changes were made.
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=${ROBOT_IP:-XXX.XXX.X.XXX} launch_rviz:=false initial_joint_controller:=joint_trajectory_controller`
Afterwards, we run this command
`ros2 launch custom_package ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true`

### Expected behaviour
The intended output is for RVIZ to open and show the current pose of the robot. The terminal outputs this error, opens RVIZ, and the robot lays flat,
`panda = MoveItPy(node_name="moveit_py_planning_scene")
ur_moveit-1  | [moveit_py.py-2]             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ur_moveit-1  | [moveit_py.py-2] RuntimeError: Planning plugin name is empty or not defined in namespace 'ompl'. Please choose one of the available plugins: chomp_interface/CHOMPPlanner, ompl_interface/OMPLPlanner, pilz_industrial_motion_planner/CommandPlanner, planning_pipeline_test/DummyPlannerManager, stomp_moveit/StompPlanner`

### Backtrace or Console output
The files I used (moveit_py.py, ur_moveit.launch.py, moveit_cpp.yaml) are [linked here](https://gist.github.com/elsa9334/4c0cb20ddbe1083e51a30ec6ae8d0add).
moveit_cpp.yaml is based on [this comment](https://github.com/moveit/moveit2_tutorials/issues/724#issuecomment-1694499977)

lmyhit, 2024-10-31:
hello, have you addressed this problem? I have the same: [moveit_py.py-2] RuntimeError: Planning plugin name is empty or not defined in namespace 'ompl'. Please choose one of the available plugins: chomp_interface/CHOMPPlanner, ompl_interface/OMPLPlanner, pilz_industrial_motion_planner/CommandPlanner, planning_pipeline_test/DummyPlannerManager, stomp_moveit/StompPlanner

elsa9334, 2024-11-01:
Hi,

unfortunately we have not. In the mean time we're still using ROS2 Humble and c++.

Pugens, 2025-02-25:
Having the same problem on Jazzy. The setup assistant  does not work out of the box and there are multiple issues, stuck here right now

lbusellato, 2025-02-26:
I had the same issue under Jazzy, it was solved by running:
```
sudo apt install ros-jazzy-moveit-*
```

It is probably overkill to install everything, but it did the trick.

JasperTan97, 2025-03-25:
Hello! 
I have got it working myself, on ROS2 Humble but using the main branch of moveit, building it from source.

This was how I passed in the config:
```python
    moveit_cpp_config = {
        "planning_scene_monitor": {
            "name": "planning_scene_monitor",
            "robot_description": "robot_description",
            "joint_state_topic": "/joint_states",
            "attached_collision_object_topic": "/moveit_cpp/planning_scene_monitor",
            "publish_planning_scene_topic": "/moveit_cpp/publish_planning_scene",
            "monitored_planning_scene_topic": "/moveit_cpp/monitored_planning_scene",
            "wait_for_initial_state_timeout": 10.0,
        },
        "planning_pipelines": {
            "pipeline_names": ["ompl"]
        },
        "plan_request_params": {
            "planning_attempts": 1,
            "planning_pipeline": "ompl",
            "max_velocity_scaling_factor": 0.2,
            "max_acceleration_scaling_factor": 0.2
        },
        "ompl": {
            "planning_plugins": ["ompl_interface/OMPLPlanner"],
            "request_adapters": ["default_planning_request_adapters/ResolveConstraintFrames",
                            "default_planning_request_adapters/ValidateWorkspaceBounds",
                            "default_planning_request_adapters/CheckStartStateBounds",
                            "default_planning_request_adapters/CheckStartStateCollision"],
            "response_adapters": ["default_planning_response_adapters/AddTimeOptimalParameterization",
                             "default_planning_response_adapters/ValidateSolution",
                             "default_planning_response_adapters/DisplayMotionPath"],
            "start_state_max_bounds_error": 0.1
        }
    }
```
I did not use the moveit config builder, but I think you can just translate it back to the yaml format. But what's important to note is that you must define the planning plugin in plural `planning_plugins`, and then put the plugin as a list. This worked for me. I hope it's not too late for you.

You have the source code [here.](https://github.com/moveit/moveit2/blob/660175c4ec8955fa99c0d24b35c7c6f1bb8c09e3/moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp#L109)

lbusellato, 2025-03-25:
Actually I ran into some other issues with it under Jazzy, and have since moved on to other stuff. While installing everything solved the problem in the issue, other errors spawned. Managed to fix most of them (without documenting it, stupidly enough), until this one:

```
[move_group.moveit.moveit.core.time_optimal_trajectory_generation]: No acceleration limit was defined for joint joint_1! You have to define acceleration limits in the URDF or joint_limits.yaml
```

I dont know when, but when I get back to trying and making moveit work under Jazzy I'll try messing around with the yaml as @JasperTan97  pointed out.

JasperTan97, 2025-03-25:
Very not relevant to the main topic now.

> [move_group.moveit.moveit.core.time_optimal_trajectory_generation]: No acceleration limit was defined for joint joint_1! You have to define acceleration limits in the URDF or joint_limits.yaml

I fixed that by passing a joint limits config into the moveit node with the following structure:
```yaml
joint_limits:
  fr3_joint1:
    has_position_limits: true
    min_position: -2.7437
    max_position:  2.7437
    has_velocity_limits: true
    max_velocity: 2.62
    has_acceleration_limits: true
    max_acceleration: 10.0
```