moveit/moveit2_tutorials #971: CI Failing due to new `scope_guard` dependency in MoveIt Task Constructor

sea-bass, 2024-09-20:
### Description

CI is failing with an error:

```
  --- stderr: moveit_task_constructor_core
  moveit_task_constructor_core: You did not request a specific build type: Choosing 'Release' for maximum performance
  /home/runner/work/moveit2_tutorials/moveit2_tutorials/.work/upstream_ws/src/moveit_task_constructor/core/src/task.cpp:49:10: fatal error: scope_guard/scope_guard.hpp: No such file or directory
     49 | #include <scope_guard/scope_guard.hpp>
        |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

See e.g., https://github.com/moveit/moveit2_tutorials/actions/runs/10917280200/job/30346870482#step:6:620

This seems to have been introduced in a merge of `master` to `ros2` which adds a new submodule. See [this commit](https://github.com/moveit/moveit_task_constructor/commit/cc7f9f0585e4e89ae1faf807a1ce180d9515d010#diff-fe7afb5c9c916e521401d3fcfb4277d5071798c3baf83baf11d6071742823584R6-R8)

@rhaschke Do you know how we ensure this submodule comes in properly? Presumably the existing `pybind11` one is already working? Thanks!

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 24.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2024-09-20:
Will have a look

rhaschke, 2024-09-20:
I think this was a cache issue. vcs fails to checkout new submodules if the package folder already exists (from cache).
I retriggered the build: https://github.com/moveit/moveit2_tutorials/actions/runs/10917280200/job/30417030494

closed 2024-09-20 by rhaschke