moveit/moveit2_tutorials #974: What is the purpose of easy binary install, if we cannot even try the tutorials with it?

Pugens, 2024-09-23:
### Description

Following a ros jazzy full desktop install and `ros-jazzy-moveit` install with `apt-get`, I cannot do any of the official introductory tutorials for moveit2. I am not an expert of ubuntu dist packages and source install and the like, so sorry from my part if I am failing to see the ovious here. But, should it not be a little bit simpler? Like, can there be at least 1 example in the tutorials repo explaining how to use the easy and quick binary install of movit2 with a simple ros2 package? My laptop is still ok, with i7, 16GB and RTX, and yet I hang and crash my Ubuntu GUI when I "have to" build the whole moveit2 source code.

I'll try to learn Movit2 from inside a container in the meanwhile... Or I could try the `--executor sequential` option, but that is not the point of my issue with this repo. Thank you for the help!

### Your environment
* ROS Distro: Jazzy
* OS Version: Ubuntu 24.04
* Binary build
* If binary, which release version? `ros-jazzy-moveit`

### Steps to reproduce
Follow these instructions:
https://moveit.ai/install-moveit2/binary/
Then the tutorials that are linked on that same page:
https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html
Uninstall the just installed binaries to build EVERYTHING from scratch

### Expected behaviour
Crash you sub-par computer

### Backtrace or Console output
Cannot report as my Ubuntu crashes (I am not trying for the 3rd time!), but all packages where building correctly as supposed, at least up to `movit_core` or something like that.

mikeferguson, 2024-12-11:
@Pittmann-XIE the sync of the latest jazzy release has not yet happened - probably a few more days before 2.12 lands in the public Debian repo.

You could try changing to the `testing` Debians if you're familiar with that workflow (if you're not, it might be best to wait for the official sync).

Pittmann-XIE, 2024-12-12:
Thanks. Does the binary installation of moveit2-jazzy supports Moveit-Python-APIs? Or only build from source supports Python APIs?

sea-bass, 2024-12-12:
The binaries should support the MoveIt Python interface as well.

mikeferguson, 2024-12-18:
With the 2.12.0 release of Debians, the tutorials do build on Jazzy (I'm tracking changes that need to be made to the tutorials as I run through testing over here: https://github.com/moveit/moveit2_tutorials/issues/995)

closed 2024-12-18 by mikeferguson