moveit/moveit2_tutorials #976: moveit_planners_ompl package failed

mjkim001130, 2024-09-29:
### Description

Hello, I have issue that following the moveit Tutorials 
When I build my Colcon workspace, 

```bash
cd ~/ws_moveit
colcon build --mixin release --executor sequential
```
And it printed the issue

```
---
Failed   <<< moveit_planners_ompl [1.03s, exited with code 2]

Summary: 26 packages finished [5.27s]
  1 package failed: moveit_planners_ompl
  1 package had stderr output: moveit_planners_ompl
  28 packages not processed
```
### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04

sea-bass, 2024-09-29:
Could you share the error message above those lines so we know what actually failed? Thanks!

mjkim001130, 2024-09-30:
Thanks for your repsones.

It has same issue with `colcon build --mixin release`

```
/usr/bin/ld: warning: libboost_serialization.so.1.74.0, needed by libmoveit_ompl_interface.so.2.5.5, may conflict with libboost_serialization.so.1.82.0
/usr/bin/ld: warning: libboost_serialization.so.1.74.0, needed by /opt/ros/humble/lib/x86_64-linux-gnu/libompl.so, may conflict with libboost_serialization.so.1.82.0
/usr/bin/ld: warning: libboost_serialization.so.1.74.0, needed by /opt/ros/humble/lib/x86_64-linux-gnu/libompl.so, may conflict with libboost_serialization.so.1.82.0
/usr/bin/ld: warning: libboost_serialization.so.1.74.0, needed by /opt/ros/humble/lib/x86_64-linux-gnu/libompl.so, may conflict with libboost_serialization.so.1.82.0
/usr/bin/ld: warning: libboost_serialization.so.1.74.0, needed by /opt/ros/humble/lib/x86_64-linux-gnu/libompl.so, may conflict with libboost_serialization.so.1.82.0
/usr/bin/ld: CMakeFiles/test_constrained_state_validity_checker.dir/test/test_constrained_state_validity_checker.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > testing::PrintToString<std::shared_ptr<ompl_interface::ModelBasedStateSpace> >(std::shared_ptr<ompl_interface::ModelBasedStateSpace> const&)':
test_constrained_state_validity_checker.cpp:(.text._ZN7testing13PrintToStringISt10shared_ptrIN14ompl_interface20ModelBasedStateSpaceEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_[_ZN7testing13PrintToStringISt10shared_ptrIN14ompl_interface20ModelBasedStateSpaceEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_]+0x1e6): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/test_state_validity_checker.dir/test/test_state_validity_checker.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > testing::PrintToString<std::shared_ptr<ompl_interface::ModelBasedPlanningContext> >(std::shared_ptr<ompl_interface::ModelBasedPlanningContext> const&)':
test_state_validity_checker.cpp:(.text._ZN7testing13PrintToStringISt10shared_ptrIN14ompl_interface25ModelBasedPlanningContextEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_[_ZN7testing13PrintToStringISt10shared_ptrIN14ompl_interface25ModelBasedPlanningContextEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_]+0x1e6): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
/usr/bin/ld: CMakeFiles/test_state_validity_checker.dir/test/test_state_validity_checker.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > testing::PrintToString<std::shared_ptr<ompl_interface::ModelBasedStateSpace> >(std::shared_ptr<ompl_interface::ModelBasedStateSpace> const&)':
test_state_validity_checker.cpp:(.text._ZN7testing13PrintToStringISt10shared_ptrIN14ompl_interface20ModelBasedStateSpaceEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_[_ZN7testing13PrintToStringISt10shared_ptrIN14ompl_interface20ModelBasedStateSpaceEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_]+0x1e6): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [ompl_interface/CMakeFiles/test_constrained_state_validity_checker.dir/build.make:354: ompl_interface/test_constrained_state_validity_checker] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:378: ompl_interface/CMakeFiles/test_constrained_state_validity_checker.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
collect2: error: ld returned 1 exit status
gmake[2]: *** [ompl_interface/CMakeFiles/test_state_validity_checker.dir/build.make:354: ompl_interface/test_state_validity_checker] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:294: ompl_interface/CMakeFiles/test_state_validity_checker.dir/all] Error 2
/usr/bin/ld: warning: libboost_serialization.so.1.74.0, needed by libmoveit_ompl_interface.so.2.5.5, may conflict with libboost_serialization.so.1.82.0
/usr/bin/ld: CMakeFiles/test_planning_context_manager.dir/test/test_planning_context_manager.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > testing::PrintToString<std::shared_ptr<ompl::geometric::SimpleSetup> >(std::shared_ptr<ompl::geometric::SimpleSetup> const&)':
test_planning_context_manager.cpp:(.text._ZN7testing13PrintToStringISt10shared_ptrIN4ompl9geometric11SimpleSetupEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_[_ZN7testing13PrintToStringISt10shared_ptrIN4ompl9geometric11SimpleSetupEEEENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_]+0x83): undefined reference to `testing::internal::PrintBytesInObjectTo(unsigned char const*, unsigned long, std::ostream*)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [ompl_interface/CMakeFiles/test_planning_context_manager.dir/build.make:354: ompl_interface/test_planning_context_manager] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:322: ompl_interface/CMakeFiles/test_planning_context_manager.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_planners_ompl [25.5s, exited with code 2]
Aborted  <<< moveit_kinematics [26.9s]                                                         
Aborted  <<< moveit_visual_tools [27.2s]                                                         
[... cut ...]
  3 packages aborted: moveit_kinematics moveit_ros_warehouse moveit_visual_tools
  16 packages had stderr output: chomp_motion_planner moveit_chomp_optimizer_adapter moveit_core moveit_kinematics moveit_planners_chomp moveit_planners_ompl moveit_resources_prbt_ikfast_manipulator_plugin moveit_ros_control_interface moveit_ros_occupancy_map_monitor moveit_ros_perception moveit_ros_planning moveit_ros_robot_interaction moveit_ros_warehouse moveit_simple_controller_manager moveit_visual_tools pilz_industrial_motion_planner_testutils
  24 packages not processed
```

sea-bass, 2024-11-23:
Hmm, as the error indicates there may be conflicting library versions of boost that are causing these linking errors. Maybe try uninstall other version of boost on your system and try building clean?

15707397862, 2025-03-26:
@mjkim001130 Hello buddy, I'm having the same problem as you, it seems like it can be fixed by changing the libboost version. But I'm worried that this will affect some of my existing works, have you solved the problem now?

da-niuma, 2025-03-26:
@mjkim001130 Hello buddy, I'm having the same problem as you right now, did you solve it?

mkygogo, 2025-04-29:
I got the same issue, any one solve it ?

ecstayalive, 2025-05-28:
For me personally, compiling packages conflicts with the anaconda environment. A brute solution is to comment out the .bashrc or .zshrc about anaconda setup. Then delete the previous compilation cache and recompile.