moveit/moveit2_tutorials #981: moveit2_tutorials demo.launch.py not working

ItsTomSav, 2024-10-22:
### Description

Launch the demo.launch.py file fails probably due to an invalid parameter

### Your environment
* ROS Distro: [Iron]
* OS Version: e.g. Ubuntu 22.04 (WSL2)
* Source or Binary build? Binary
* If binary, which release version? ros-iron-desktop 0.10.0-3

### Steps to reproduce
ros2 launch moveit2_tutorials demo.launch.py

### Expected behaviour
Demo launches

### Backtrace or Console output
[INFO] [launch]: All log files can be found below /home/tommaso/.ros/log/2024-10-22-15-00-24-589793-LAPTOP-TQEEJ2VI-1680
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: Invalid parameter "sim_gazebo"

FrankSav80, 2024-10-22:
I also have the same problem with the same configuration as you (iron) and cannot solve it.

I previously encountered issues in the tutorial [Getting Started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) at the following point:

Move into your Colcon workspace and pull the MoveIt tutorials source, where <branch> can be e.g. humble for ROS Humble, or main for the latest version of the tutorials:
```
cd ~/ws_moveit/src
git clone -b <branch> https://github.com/moveit/moveit2_tutorials 
```

Using `git clone -b main https://github.com/moveit/moveit2_tutorials` gave me problems. I solved it by running the following commands:

```
rm -rf ros2_robotiq_gripper/
git clone  https://github.com/PickNikRobotics/ros2_robotiq_gripper.git  --branch humble
```

Perhaps this is where the current problem stems from?

mikeferguson, 2024-12-18:
Iron is now EOL

closed 2024-12-18 by mikeferguson