moveit/moveit2_tutorials #986: 403 Forbidden when trying to run Docker image

nirbheeksetia, 2024-11-12:
**Description**
I am trying to follow the official MoveIt installation tutorial from the website. (For further reference: https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html ) While running the last command I get an error concerning the architecture of my system.

**Environment**
      -ROS Distro: Jazzy 
      -OS Version: Ubuntu 24.04 (WSL2)
      - CPU information:
      [image]

**Steps to reproduce**
After following through the whole tutorial, run the final command:
```
cd ~/ws_moveit
colcon build --mixin release
```

**Error/Output**
[image]

Is there really no arm support for MoveIt? if yes, how do i fix this issue?

sea-bass, 2024-11-14:
That seems related to the `ros2_kortex` repo.

You could maybe try this PR branch, which adds `aarch64` support?
https://github.com/Kinovarobotics/ros2_kortex/pull/238

nirbheeksetia, 2024-11-15:
Allright, I will try that. Thank you for your answer!

mikeferguson, 2024-11-15:
Agree - this appears to be a ros2_kortex issue, not MoveIt itself

closed 2024-11-15 by mikeferguson