moveit/moveit2_tutorials #987: Error in Motion Planning Python API: plan_result = planning_component.plan()

Pittmann-XIE, 2024-11-13:
Hey, I am following this tutorial "https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html#getting-started" and encountered the following error.
# Problem
I ran the first part of this tutorial successfully:
`ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py`

Then I killed this process and tried to do the second part. As I ran the command:
`ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py example_file:=motion_planning_python_api_planning_scene.py`, I received this error:
`[INFO] [1731500363.915239993] [moveit_py_planning_scene]: MoveItPy instance created
[motion_planning_python_api_planning_scene.py-1] [INFO] [1731500363.915632586] [moveit_py_planning_scene]: Planning trajectory
[motion_planning_python_api_planning_scene.py-1] Traceback (most recent call last):
[motion_planning_python_api_planning_scene.py-1]   File "/home/ibrahim-hhi/ws_2/ws_ros_rolling/ws_moveit/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_planning_scene.py", line 144, in <module>
[motion_planning_python_api_planning_scene.py-1]     main()
[motion_planning_python_api_planning_scene.py-1]   File "/home/ibrahim-hhi/ws_2/ws_ros_rolling/ws_moveit/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_planning_scene.py", line 97, in main
[motion_planning_python_api_planning_scene.py-1]     plan_and_execute(panda, panda_arm, logger, sleep_time=3.0)
[motion_planning_python_api_planning_scene.py-1]   File "/home/ibrahim-hhi/ws_2/ws_ros_rolling/ws_moveit/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_planning_scene.py", line 26, in plan_and_execute
[motion_planning_python_api_planning_scene.py-1]     plan_result = planning_component.plan()
[motion_planning_python_api_planning_scene.py-1]                   ^^^^^^^^^^^^^^^^^^^^^^^^^
[motion_planning_python_api_planning_scene.py-1] TypeError: plan(): incompatible function arguments. The following argument types are supported:
[motion_planning_python_api_planning_scene.py-1]     1. (self: moveit.planning.PlanningComponent, single_plan_parameters: moveit.planning.PlanRequestParameters = None, multi_plan_parameters: moveit.planning.MultiPipelinePlanRequestParameters = None, planning_scene: planning_scene::PlanningScene = None, solution_selection_function: Optional[Callable[[List[planning_interface::MotionPlanResponse]], planning_interface::MotionPlanResponse]] = None, stopping_criterion_callback: Optional[Callable[[moveit::planning_pipeline_interfaces::PlanResponsesContainer, List[moveit_msgs::msg::MotionPlanRequest_<std::allocator<void> >]], bool]] = None) -> planning_interface::MotionPlanResponse
[motion_planning_python_api_planning_scene.py-1] 
[motion_planning_python_api_planning_scene.py-1] Invoked with: <moveit.planning.PlanningComponent object at 0x70389f4438b0>
[motion_planning_python_api_planning_scene.py-1] 
[motion_planning_python_api_planning_scene.py-1] Did you forget to `#include <pybind11/stl.h>`? Or <pybind11/complex.h>,
[motion_planning_python_api_planning_scene.py-1] <pybind11/functional.h>, <pybind11/chrono.h>, etc. Some automatic
[motion_planning_python_api_planning_scene.py-1] conversions are optional and require extra headers to be included
[motion_planning_python_api_planning_scene.py-1] when compiling your pybind11 module.
[motion_planning_python_api_planning_scene.py-1] [INFO] [1731500363.926675826] [moveit_487904131.moveit.ros.moveit_cpp]: Deleting MoveItCpp
[rviz2-4] [INFO] [1731500363.931211563] [rviz2.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[rviz2-4] [INFO] [1731500363.932542608] [rviz2.moveit.ros.planning_scene_monitor]: Listening to '/moveit_cpp/monitored_planning_scene'
[INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process has finished cleanly [pid 260732]
[motion_planning_python_api_planning_scene.py-1] [INFO] [1731500363.940214900] [moveit_487904131.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[motion_planning_python_api_planning_scene.py-1] [INFO] [1731500363.941666926] [moveit_487904131.moveit.ros.planning_scene_monitor]: Stopping world geometry monitor
[motion_planning_python_api_planning_scene.py-1] [INFO] [1731500363.942018265] [moveit_487904131.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process has finished cleanly [pid 260734]
[ERROR] [motion_planning_python_api_planning_scene.py-1]: process has died [pid 260727, exit code 1, cmd '/home/ibrahim-hhi/ws_2/ws_ros_rolling/ws_moveit/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_planning_scene.py --ros-args -r __node:=moveit_py --params-file /tmp/launch_params_ac99mpn6'].
[rviz2-4] [INFO] [1731500368.935666663] [rviz2.moveit.ros.planning_scene_monitor]: GetPlanningScene service call to get_planning_scene timed out. at /home/ibrahim-hhi/ws_2/ws_ros_rolling/ws_moveit/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:618`

Does anyone know how to fix this problem? Thanks in advance.

# System:
ubuntu 24.04
Ros version: ros-rolling

Lydaidai, 2024-12-04:
Same env and same problem, is there any update? Looks like a compatibility issue with the Python API.

Lydaidai, 2024-12-04:
Hi, I solved this by adding `from moveit.core.planning_scene import PlanningScene`.