moveit/moveit2_tutorials #992: Getting Started says ROS2 Iron is supported - but there's no iron branch?

dghw, 2024-11-21:
Did there used to be an Iron branch for this repo? I'm trying to fix some bugs in my code that uses the Python motion planning API, so was trying to take a look again at the tutorials. However:

1. MoveIt 2 Humble doesn't support the Python motion planning API
2. MoveIt 2 Iron does support it, but there's no Iron version of this repo. I tried with main but got a lot of errors relating to mismatches between ros-iron-ros2-control and the rolling ros-rolling-robotiq-driver that gets installed
3. MoveIt2 Jazzy is not really an option for me at the moment due to the need to update the OS 

I know that ROS2 Iron is unsupported soon (already?). Am I just out of luck here?  

It's a little confusing actually because the Getting Started page on the MoveIt 2 tutorials clearly says Iron is supported, but if you follow the instructions through you'll eventually reach an error when you try to pull the iron branch of this repo, which doesn't exist as far as I can see.

sea-bass, 2024-11-23:
You're right -- an `iron`  branch of this repository was never created, although there is an `iron` branch of the `moveit2` repository, at least.

Since no more changes are planned to Iron at this point, it could be possible to work on a "working" set of tutorials here, but it would take effort from the community.

mikeferguson, 2024-11-28:
Iron is pretty much EOL - last sync has happened. I just opened a PR to remove the Iron references (and add jazzy references) to the getting started page

closed 2024-11-28 by mikeferguson