moveit/moveit2_tutorials #995: Testing tutorials on Jazzy

mikeferguson, 2024-12-18:
I'm testing things out, trying to run from Debians to see if we can switch the install docs. Notes:

 * Switched to ros-testing - to get Moveit2 version 2.12.0 (`sudo apt install ros-jazzy-moveit`)
 * Had to build moveit_task_constructor `ros2` branch - since it's not available in Debians yet for Jazzy?
 * Did the rosdep command to install dependencies
 * Workspace then built successfully

sea-bass, 2024-12-18:
I don't think MoveIt Task Constructors has any debians for ROS 2:

https://index.ros.org/search/?term=moveit_task_constructor

mikeferguson, 2024-12-18:
Huh - the kinova packages are in `rolling` as debians - but not jazzy...

sea-bass, 2024-12-18:
I think the package index is lying to us... looking at the rosdistro PRs, this information is consistent with your findings.

https://github.com/ros/rosdistro/pulls?q=is%3Apr+ros2_kortex

No PRs since 2023.

I'm guessing my standing PR (https://github.com/Kinovarobotics/ros2_kortex/pull/228) among other things, is still blocking them from actually releasing to 24.04 / the post Gazebo classic world.

mikeferguson, 2024-12-18:
> BTW that branch/fork of `ros2_kortex` is what this repo is currently using.

Yours of course! And it works with that - fails with mainline...

mikeferguson, 2024-12-18:
Alright, once 2.12.x lands in debians, I suggest we:

 * Add a moveit2_tutorials.jazzy.repos, with just:
   * moveit_task_constructor
   * the branched ros2_kortex
   * ros2_robotiq_gripper
   * serial (also a random branch)
 * Update the getting started page to NOT remove moveit from debians, and just build the above

gavanderhoorn, 2024-12-18:
> I think the package index is lying to us...

For checking state of (released) packages, I'd only check what [repo.ros2.org/status_page](https://repo.ros2.org/status_page/) says.

For the Kinova pkgs:

- [status_page/ros_rolling_default.html?q=kinova](https://repo.ros2.org/status_page/ros_rolling_default.html?q=kinova)
- [status_page/ros_jazzy_default.html?q=kinova](https://repo.ros2.org/status_page/ros_jazzy_default.html?q=kinova)

Task constructor:

- [ros_rolling_default.html?q=moveit_t](https://repo.ros2.org/status_page/ros_rolling_default.html?q=moveit_t)
- [status_page/ros_jazzy_default.html?q=moveit_t](https://repo.ros2.org/status_page/ros_jazzy_default.html?q=moveit_t)