moveit/moveit2_tutorials #998: Controller manager error -[WARN] [1735037100.684428992] [spawner_panda_arm_controller]: Could not contact service /controller_manager/list_controllers

Bharath21234, 2024-12-25:
I am trying to visualise a 3dof robotic arm and control this using my keyboard. I am on ROS2 Humble Ubuntu 22.04 and am using this moveit2 tutorial for controlling it : https://github.com/moveit/moveit2_tutorials/tree/main/doc/examples/realtime_servo https://moveit.picknik.ai/humble/doc/examples/realtime_servo/realtime_servo_tutorial.html#using-the-c-interface

When I run my launch file, I can see the arm in Rviz, but it its not moving when I run

ros2 run moveit2_tutorials servo_keyboard_input

in a new terminal. Im guessing its because of this error that Im getting : [spawner-5] [WARN] [1735037100.684428992] [spawner_panda_arm_controller]: Could not contact service /controller_manager/list_controllers

This is my launch file

import os
import launch_ros
import launch
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.conditions import IfCondition, UnlessCondition
from launch.substitutions import LaunchConfiguration
from ament_index_python.packages import get_package_share_directory
from launch.actions import ExecuteProcess
from moveit_configs_utils import MoveItConfigsBuilder
from launch_param_builder import ParameterBuilder

def generate_launch_description():
    moveit_config = (
        MoveItConfigsBuilder(robot_name = "omnipointer", package_name = "my_robot_description")
        .robot_description(file_path="urdf/three_dof_arm.urdf")
        .robot_description_semantic(file_path="urdf/srdf.srdf")
        .robot_description_kinematics(file_path = "moveit_config/kinematics.yaml")
        .joint_limits(file_path="moveit_config/joint_limits.yaml")
        .pilz_cartesian_limits(file_path = "moveit_config/pilz_cartesian_limits.yaml")
        
        .to_moveit_configs()
    )
    # Launch Servo as a standalone node or as a "node component" for better latency/efficiency
    launch_as_standalone_node = LaunchConfiguration(
        "launch_as_standalone_node", default="false"
    )
    # Get parameters for the Servo
    servo_params = {
        "moveit_servo": ParameterBuilder("my_robot_description")
        .yaml("moveit_config/servo_simulated_config.yaml")
        .to_dict()
    }
    # This sets the update rate and planning group name for the acceleration limiting filter.
    planning_group_name = {"planning_group_name": "manipulator"}

    # RViz
    rviz_config_file = (
        get_package_share_directory("my_robot_description")
        + "/moveit_config/rviz_config.rviz"
    )
    rviz_node = launch_ros.actions.Node(
        package="rviz2",
        executable="rviz2",
        name="rviz2",
        output="log",
        arguments=["-d", rviz_config_file],
        parameters=[
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
        ],
    )

    robot_publisher_node = launch_ros.actions.Node(
        package="robot_state_publisher",
        executable="robot_state_publisher",
        name="robot_state_publisher",
        parameters=[{"robot_description": moveit_config.robot_description}],
        output="screen",
)
 # ros2_control using FakeSystem as hardware
    ros2_controllers_path = os.path.join(
        get_package_share_directory("my_robot_description"),
        "moveit_config",
        "ros2_controllers.yaml",
    )
    ros2_control_node = launch_ros.actions.Node(
        package="controller_manager",
        namespace = '/',
        executable="ros2_control_node",
        parameters=[ros2_controllers_path],
        remappings=[
            ("/controller_manager/robot_description", "/robot_description"),
        ],
        output="screen",
    )

    joint_state_broadcaster_spawner = launch_ros.actions.Node(
        package="controller_manager",
        namespace = "/",
        executable="spawner",
        arguments=[
            "joint_state_broadcaster",
            "--controller-manager-timeout",
            "300",
            "--controller-manager",
            "/controller_manager",
        ],
    )
    panda_arm_controller_spawner = launch_ros.actions.Node(
        package="controller_manager",
        executable="spawner",
        arguments=["panda_arm_controller", "-c", "/controller_manager"],
    )
    # Launch as much as possible in components
    container = launch_ros.actions.ComposableNodeContainer(
        name="moveit_servo_demo_container",
        namespace = "/",
        package="rclcpp_components",
        executable="component_container_mt",
        composable_node_descriptions=[
            # Example of launching Servo as a node component
            # Launching as a node component makes ROS 2 intraprocess communication more efficient.
            launch_ros.descriptions.ComposableNode(
                package="moveit_servo",
                plugin="moveit_servo::ServoNode",
                name="servo_node",
                parameters=[
                    servo_params,
                    planning_group_name,
                    moveit_config.robot_description,
                    moveit_config.robot_description_semantic,
                    moveit_config.robot_description_kinematics,
                ],
                condition=UnlessCondition(launch_as_standalone_node),
            ),
            launch_ros.descriptions.ComposableNode(
                package="tf2_ros",
                plugin="tf2_ros::StaticTransformBroadcasterNode",
                name="static_tf2_broadcaster",
                parameters=[{"child_frame_id": "base_link", "frame_id": "world"}],
            ),
        ],
        output="screen",
    )

    # Servo control node
    servo_node = Node(
        package="moveit2_tutorials",  # Ensure this package name is correct
        executable="servo_cpp_interface_demo",  # Check that this executable exists
        output="screen",
        parameters=[
            servo_params,
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
        ],
    )

    demo_node = launch_ros.actions.Node(
        package="my_robot_description",
        executable="servo_control",
        name="servo_control",
        output="screen",
    )

    return launch.LaunchDescription(
        [
            rviz_node,
            robot_publisher_node,
            ros2_control_node,
            joint_state_broadcaster_spawner,
            panda_arm_controller_spawner,
            servo_node,
            container,
            launch.actions.TimerAction(period=350.0, actions=[demo_node]),
        ]
    )

This is my ros2_controllers.yaml file :

# This config file is used by ros2_control
controller_manager:
  ros__parameters:
    update_rate: 1000  # Hz

    panda_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

panda_arm_controller:
  ros__parameters:
    joints:
      - arm_joint_1, 
      - arm_joint_2,
      - arm_joint_3,
      - arm_joint_4,
      - arm_joint_5
    command_interfaces:
      - position
    state_interfaces:
      -position 
      -velocity

Any way to solve this? When I run ros2 node list while the launch file is running, the controller_manager does show up so that node is running. Don't know what is causing this error. Another thing I noticed is that the robot_state_publisher node only shows up for about 10 seconds when I run ros2 node list after running the launch file. Could this be the reason for this issue related to the controller manager or is it a seperate problem? Lastly, I tried only visualising the file and controlling the angles using the slider of joint_state_publisher_gui and this works perfectly well.

sea-bass, 2024-12-26:
There are quite a few error messages that are taking down nodes, which you may want to resolve one at a time.

The robot state publisher seems to be failing here:

```
[robot_state_publisher-2] terminate called after throwing an instance of 'std::runtime_error'
[robot_state_publisher-2] what(): robot_description parameter must not be empty
```

The waiting for the controller manager likely fails because of this error that takes it down:

```
controller_manager]: Received robot description file.
[ros2_control_node-3] [ERROR] [1735146709.843205770] [controller_manager]: The published robot description file (urdf) seems not to be genuine. The following error was caught:no ros2_control tag
```

MoveIt Sero is using a smoothing plugin that is not available on Humble.

```
Exception while loading the smoothing plugin 'online_signal_smoothing::AccelerationLimitedPlugin': 'According to the loaded plugin descriptions the class online_signal_smoothing::AccelerationLimitedPlugin with base class type online_signal_smoothing::SmoothingBaseClass does not exist. Declared types are online_signal_smoothing::ButterworthFilterPlugin'
```

Lastly, there may be other misconfigurations in the robot model:

```
[moveit_robot_model.robot_model]: The model for joint 'arm_joint_5' requires 0 variable values, but only 1 variable values were supplied in default state 'home' for group 'manipulator'
```

Bharath21234, 2024-12-27:
Thank you for the prompt response. I have managed to fix the first two errors. How do I change the smoothing plugin to ButterworthFIlterPlugin? Thanks in advance

sea-bass, 2024-12-28:
Good question. I could not find good documentation for this. 

You're using Humble, right?

Basically, you can start by looking at this file.

https://github.com/moveit/moveit2/blob/f1937a0caa0f30188bb6944486318d98cf09c80d/moveit_ros/moveit_servo/launch/servo_example.launch.py#L43

Notice that this points to a YAML with all the parameter descriptions for launching MoveIt Servo. The plugin name can then be changed here:

https://github.com/moveit/moveit2/blob/f1937a0caa0f30188bb6944486318d98cf09c80d/moveit_ros/moveit_servo/config/panda_simulated_config.yaml#L32

---

The Acceleration smoothing plugin is only available on newer versions of MoveIt 2, which you can get either by upgrading to a new ROS version and using the binaries... or building the `main` branch of MoveIt 2 from source which should still work.

Bharath21234, 2025-01-01:
Okay this works! Thank you! one more issue im currently facing is that it gives this error now when I run the launch file ` [servo_cpp_interface_demo-6] [ERROR] [1735720202.055959204] [moveit_robot_model.robot_model]: Link 'panda_link8' not found in model 'omnipointer'
[servo_cpp_interface_demo-6] [ERROR] [1735720202.056024266] [moveit_exceptions.exceptions]: Invalid link
[servo_cpp_interface_demo-6] Exception thrown.
[servo_cpp_interface_demo-6] terminate called after throwing an instance of 'moveit::Exception'
[servo_cpp_interface_demo-6]   what():  Invalid link
[ERROR] [servo_cpp_interface_demo-6]: process has died [pid 17037, exit code -6, cmd '/home/bharath/ws_moveit2/install/moveit2_tutorials/lib/moveit2_tutorials/servo_cpp_interface_demo --ros-args --params-file /tmp/launch_params_4vuftdry --params-file /tmp/launch_params_6aiumkcm --params-file /tmp/launch_params_993uh3zf'].`So I searched for panda_link8 in my whole workspace and it isnt referenced anywhere. Would I need to change the original config file inside ws_moveit2 ? Because my urdf doesnt have 8 links. It only has 5. But changing it in the original config file might cause an issue later on right ? Also, I searched for the servo_cpp_interface_demo file in the ws_moveit2 and Im not able to find it

sea-bass, 2025-01-01:
It likely comes from here: https://github.com/moveit/moveit2/blob/f1937a0caa0f30188bb6944486318d98cf09c80d/moveit_ros/moveit_servo/config/panda_simulated_config.yaml#L45

So yes, you should either modify this file for your robot model and its groups/links, or you make a copy of it and reference that.

Bharath21234, 2025-01-01:
Ohh alright okay that resolves the error. But now its showing the same for panda_link0 : `[servo_cpp_interface_demo-6] [ERROR] [1735744535.337608595] [moveit_robot_model.robot_model]: Link 'panda_link0' not found in model 'omnipointer'
[servo_cpp_interface_demo-6] [ERROR] [1735744535.337693789] [moveit_exceptions.exceptions]: Invalid link
[servo_cpp_interface_demo-6] Exception thrown.
[servo_cpp_interface_demo-6] terminate called after throwing an instance of 'moveit::Exception'
[servo_cpp_interface_demo-6]   what():  Invalid link
[ERROR] [servo_cpp_interface_demo-6]: process has died [pid 120808, exit code -6, cmd '/home/bharath/ws_moveit2/install/moveit2_tutorials/lib/moveit2_tutorials/servo_cpp_interface_demo --ros-args --params-file /tmp/launch_params_77hneitz --params-file /tmp/launch_params_tyr_3ccx --params-file /tmp/launch_params_lc634ri9'].` I edited the panda_simulated_config.yaml file to change the name of the link from panda_link0 but this error still shows up. 
Also on a side note, why are there two yaml files inside the config folder? Do I need both of them or just one? Because currently I only have the panda_simulated_config one.....

sea-bass, 2025-01-01:
Did you rebuild after making more modifications? (Or are you using --symlink-install?)

Only one YAML file is needed, which is the one you are using in your launch file. Think of it as different sets of parameters for different robots/applications.