moveit/moveit2_tutorials #1000: Proper initialization of `std::shared_ptr<moveit_cpp::MoveItCpp>`

Jakeisthesnake, 2025-01-03:
I am trying copy the demo for moveit_drake (a GSoC project from 2024 that built an adapter to give Moveit access to some Drake functionality). I started with the motion planning pipeline tutorial and first switched from a panda arm to a pendulum. However, when I tried to initialize a `std::shared_ptr<moveit_cpp::MoveItCpp> ` object, the code hangs on the follow line:

in 
```
bool MoveItCpp::loadPlanningSceneMonitor(const PlanningSceneMonitorOptions& options)
{
  [. . .]
    return planning_scene_monitor_->getStateMonitor()->waitForCurrentState(node_->now(), options.wait_for_initial_state_timeout); 
  [. . .]
}
```

Looking at the code in current_state_monitor.cpp, it seems that this must be a deadlock issue on the following line:

```
bool CurrentStateMonitor::waitForCurrentState(const rclcpp::Time& t, double wait_time_s) const
{
  [. . .]
      state_update_condition_.wait_for(lock, (timeout - elapsed).to_chrono<std::chrono::duration<double>>());
  [. . .]
}
```

I found the following, possibly related issue, but I wasn't sure how to apply it to my problem:
https://github.com/moveit/moveit2/issues/496

Any help would be appreciated.

Running Moveit2 as built from source in July on top of ros2 Iron.
Here is a link to my:

launch file
https://github.com/Jakeisthesnake/Moveit_drake_adapter/blob/main/src/moveit2_tutorials/motion_planning_pipeline/launch/pendulum_motion_planning_pipeline_tutorial.launch.py

src file
https://github.com/Jakeisthesnake/Moveit_drake_adapter/blob/main/src/moveit2_tutorials/motion_planning_pipeline/src/pendulum_motion_planning_pipeline_tutorial.cpp

and here is the full log output:
```
jake@jake-Blade:~/moveit2_ws_main$ ros2 launch moveit2_tutorials pendulum_motion_planning_pipeline_tutorial.launch.py 
[INFO] [launch]: All log files can be found below /home/jake/.ros/log/2025-01-02-17-22-22-075495-jake-Blade-7240
[INFO] [launch]: Default logging verbosity is set to INFO
/home/jake/moveit2_ws_main/install/moveit_configs_utils/share/moveit_configs_utils/default_configs/ompl_planning.yaml
/home/jake/moveit2_ws_main/install/moveit_configs_utils/share/moveit_configs_utils/default_configs/stomp_planning.yaml
[INFO] [rviz2-1]: process started with pid [7243]
[INFO] [pendulum_motion_planning_pipeline_tutorial-2]: process started with pid [7245]
[INFO] [static_transform_publisher-3]: process started with pid [7247]
[static_transform_publisher-3] [WARN] [1735867342.368248470] []: Old-style arguments are deprecated; see --help for new-style arguments
[rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[static_transform_publisher-3] [INFO] [1735867342.395201145] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'world' to 'base_link'
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.462256774] [pendulum_motion_planning_pipeline]: HEREEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE main
[pendulum_motion_planning_pipeline_tutorial-2] 
[pendulum_motion_planning_pipeline_tutorial-2] [WARN] [1735867342.463503524] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.479497193] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.rdf_loader]: Loaded robot model in 0.0098873 seconds
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.479627369] [pendulum_motion_planning_pipeline_tutorial.moveit.core.robot_model]: Loading robot model 'jakes_pendulum'...
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.492456722] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]: Kinematics solver groups:
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.492508901] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]:  - pendulum_arm
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.492517708] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]: Robot Model Information:
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.492524587] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]:  - Joint Model Group: end_ball
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.492530895] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]:  - Joint Model Group: pendulum_arm
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.495561635] [pendulum_motion_planning_pipeline_tutorial.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'pendulum_arm': 1
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.538030405] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.538186594] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.moveit_cpp]: Listening to 'joint_states' for joint states
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.539247649] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.540165538] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.540195912] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.planning_scene_monitor]: Stopping existing planning scene publisher.
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1735867342.540356545] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[... 43 lines cut ...]
[rviz2-1] [INFO] [1735867411.262326658] [rviz2.moveit.ros.move_group_interface]: Ready to take commands for planning group end_ball.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[static_transform_publisher-3] [INFO] [1735869047.600439402] [rclcpp]: signal_handler(signum=2)
[rviz2-1] [INFO] [1735869047.600467677] [rclcpp]: signal_handler(signum=2)
[rviz2-1] [WARN] [1735869047.602066191] [interactive_marker_display_102930516741856]: Server not available while running, resetting
[rviz2-1] [ERROR] [1735869047.659188131] [rviz2.moveit.ros.background_processing]: Exception caught while processing action 'Frame::changePlanningGroup': std::future_error: Broken promise
[rviz2-1] [INFO] [1735869047.670961723] [rviz2.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor
[INFO] [static_transform_publisher-3]: process has finished cleanly [pid 7247]
[rviz2-1] 
[rviz2-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[rviz2-1] This error state is being overwritten:
[rviz2-1] 
[rviz2-1]   'rcl node's context is invalid, at /home/jake/ros2_iron/src/ros2/rcl/rcl/src/rcl/node.c:446'
[rviz2-1] 
[rviz2-1] with this new error message:
[rviz2-1] 
[rviz2-1]   'cannot publish data, at /home/jake/ros2_iron/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:62'
[rviz2-1] 
[rviz2-1] rcutils_reset_error() should be called after error handling to avoid this.
[rviz2-1] <<<
[rviz2-1] cannot publish data, at /home/jake/ros2_iron/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:62 during '__function__'
[rviz2-1] Fail in delete datareader, at /home/jake/ros2_iron/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_service.cpp:100 during '__function__'
[ERROR] [rviz2-1]: process has died [pid 7243, exit code -11, cmd '/home/jake/ros2_iron/install/rviz2/lib/rviz2/rviz2 -d /home/jake/moveit2_ws_main/install/moveit_drake/share/moveit_drake/config/pendulum_moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_p4ojwavq --params-file /tmp/launch_params_i03nyhqu --params-file /tmp/launch_params_t9tu2udv --params-file /tmp/launch_params_89ets3sj --params-file /tmp/launch_params_0alwngu_'].

```

Jakeisthesnake, 2025-01-03:
I thought it might make sense to post in moveit2_tutorials since I figured that I could take what I learn here and add it to a (new?) tutorial. Though, I suppose the more likely option is that I missed something that the tutorials already cover.

Jakeisthesnake, 2025-01-03:
Currently, I am looking to see what needs to be initialized in order to update the state (which would then unblock waitForCurrentState).

Jakeisthesnake, 2025-01-03:
Ah, running the working moveit_drake demo gives the following logs, which suggests that I should look into what calls updateMulitDofJoints and that I should make sure my joint_state_broadcaster is working.
```

[pipeline_testbench_example-4] [INFO] [1735934567.991662476] [pipeline_testbench_example.moveit.ros.current_state_monitor]: updateMultiDofJoints LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934567.991982525] [pipeline_testbench_example.moveit.ros.current_state_monitor]: updateMultiDofJoints LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934567.994099900] [pipeline_testbench_example.moveit.ros.current_state_monitor]: waitForCurrentState LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934567.994141874] [pipeline_testbench_example.moveit.ros.current_state_monitor]: waitFor current_state_time
[pipeline_testbench_example-4] [INFO] [1735934567.994155798] [pipeline_testbench_example.moveit.ros.current_state_monitor]: CHHHHHHEEEEECCKKK 2
[rviz2-3] [INFO] [1735934568.180687636] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1735934568.180921578] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1735934568.204544403] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-5] [INFO] [1735934568.312934003] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-6] [INFO] [1735934568.388549317] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-5] [INFO] [1735934568.390066287] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-5] [INFO] [1735934568.390302457] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[pipeline_testbench_example-4] [INFO] [1735934568.422104550] [pipeline_testbench_example.moveit.ros.current_state_monitor]: jointStateCallback LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934568.422228728] [pipeline_testbench_example.moveit.ros.current_state_monitor]: haveCompleteStateHelper LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934568.422259509] [pipeline_testbench_example.moveit.ros.current_state_monitor]: getCurrentStateTime LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934568.422301912] [pipeline_testbench_example.moveit.ros.current_state_monitor]: setToCurrentState LOOOOOOCK
[pipeline_testbench_example-4] [INFO] [1735934568.422371779] [pipeline_testbench_example.moveit.ros.current_state_monitor]: EEEELLLLLLLAAAAAAPPPPPSSSEEEEDDlapsed: 0.428254       requested: 1735934567.994095        lastest state time: 1735934568.421195

```

Jakeisthesnake, 2025-01-03:
So `  updateMultiDofJoints();` is called by `transformCallback()`, so what calls that? Ah, `stateStateMonitor` but only given certain conditions. Let's see if I am even calling startStateMonitor.

sea-bass, 2025-01-05:
It seems from your code snippet that you've commented out the starting of an executor spinning thread in the background: https://github.com/moveit/moveit2_tutorials/blob/b72a00c69c62d85ea25ff30cc480e0dd98567f6b/doc/examples/moveit_cpp/src/moveit_cpp_tutorial.cpp#L34

I believe you will likely need that uncommented, as you need a ROS executor to be running to process anything in ROS, such as calling services, getting topic subscriber callbacks, etc.

---

Also unrelated, but you mention ROS 2 Iron which is officially no longer supported since last November. You may want to either go back to Humble (if on 22.04), or upgrade to Ubuntu 24.04 and choose Jazzy or Rolling.

sea-bass, 2025-01-05:
Oh! I just read in more detail -- it's yelling about being unable to publish data?

Could you try commenting out these lines of yours and rebuilding?
https://github.com/Jakeisthesnake/Moveit_drake_adapter/blob/a5ee537585a9b12bac3dd4c4a657e9e7c5282d72/src/moveit2_tutorials/motion_planning_pipeline/launch/pendulum_motion_planning_pipeline_tutorial.launch.py#L152-L154

sea-bass, 2025-01-05:
Another thing you might want to try is switching the middleware to CycloneDDS to see how that goes.

```bash
sudo apt update
sudo apt install ros-iron-rmw-cyclonedds-cpp

# If this works, you can set this permanently in your ~/.bashrc
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

Jakeisthesnake, 2025-01-06:
Thanks for the replies!

1) I'll go back to Humble, but I'll do that later today. So the following results are still on Iron.

2) Well, I originally thought it was a deadlock issue. Now I think that I am not initializing something that updates currentStateMonitor (see "4)" below). I commented out those lines, rebuilt, and got the following output. (As far as I can tell, nothing changed.) 

update moveit_config with commented out lines:
```
moveit_config = (
        MoveItConfigsBuilder("jakes_pendulum")
        .robot_description(file_path="config/jakes_pendulum.urdf.xacro")
        # .planning_scene_monitor(
        #     publish_robot_description=True, publish_robot_description_semantic=True
        # )
        .planning_pipelines(pipelines=["ompl", "stomp"])
        .moveit_cpp(
            os.path.join(
                get_package_share_directory("moveit2_tutorials"),
                "config",
                "pendulum_motion_planning_tutorial.yaml",
            )
        )
        .to_moveit_configs()
    )
```
output:

```
jake@jake-Blade:~/moveit2_ws_main$ ros2 launch moveit2_tutorials pendulum_motion_planning_pipeline_tutorial.launch.py 
[INFO] [launch]: All log files can be found below /home/jake/.ros/log/2025-01-06-08-00-25-367602-jake-Blade-42893
[INFO] [launch]: Default logging verbosity is set to INFO
/home/jake/moveit2_ws_main/install/moveit_configs_utils/share/moveit_configs_utils/default_configs/ompl_planning.yaml
/home/jake/moveit2_ws_main/install/moveit_configs_utils/share/moveit_configs_utils/default_configs/stomp_planning.yaml
[INFO] [pendulum_motion_planning_pipeline_tutorial-1]: process started with pid [42896]
[INFO] [static_transform_publisher-2]: process started with pid [42898]
[static_transform_publisher-2] [WARN] [1736179225.830386771] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [INFO] [1736179225.870654244] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179225.950645052] [pendulum_motion_planning_pipeline]: HEREEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE main
[pendulum_motion_planning_pipeline_tutorial-1] 
[pendulum_motion_planning_pipeline_tutorial-1] [WARN] [1736179225.952107034] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179225.985117470] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.rdf_loader]: Loaded robot model in 0.0139121 seconds
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179225.985297554] [pendulum_motion_planning_pipeline_tutorial.moveit.core.robot_model]: Loading robot model 'jakes_pendulum'...
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.003436398] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]: Kinematics solver groups:
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.003502454] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]:  - pendulum_arm
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.003511972] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]: Robot Model Information:
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.003519102] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]:  - Joint Model Group: end_ball
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.003525373] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]:  - Joint Model Group: pendulum_arm
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.006775698] [pendulum_motion_planning_pipeline_tutorial.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'pendulum_arm': 1
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.063960568] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.064123073] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.moveit_cpp]: Listening to 'joint_states' for joint states
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.064190602] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.current_state_monitor]: STTAAAAARTstatemonitor
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.065434142] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.current_state_monitor]: tf_buffer_
[pendulum_motion_planning_pipeline_tutorial-1] [INFO] [1736179226.065471970] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[... cut ...]
[spawner-3] [WARN] [1736180344.676058946] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers
[spawner-3] [INFO] [1736180344.677982624] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[spawner-3] [WARN] [1736180354.702627431] [spawner_joint_state_broadcaster]: Could not contact service /controller_manager/list_controllers

```
so now I am trying to figure out why the joint_state_broadcaster can't contact the service.

Jakeisthesnake, 2025-01-06:
Woo! I think we can close the issue. I needed to add the ros2_control_node. I'm not sure how exactly I would prevent this for other people with more documentation. Maybe some sort of glossary of useful nodes to have in a launch file? If you think that is helpful, I can start working on it. Otherwise we can close the issue.

Read here (https://answers.ros.org/question/412650/) that the reason for `Could not contact service /controller_manager/list_controllers` might be that I don't have the ros2_control_node in the launch file. Added it and now I am getting the explicit message `Failed loading controller joint_state_broadcaster`

Added to launch file:
```
ros2_controllers_path = os.path.join(
        get_package_share_directory("jakes_pendulum_moveit_config"),
        "config",
        "ros2_controllers.yaml",
    )

    ros2_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[moveit_config.robot_description, ros2_controllers_path],
        output="screen",
    )
```

```
jake@jake-Blade:~/moveit2_ws_main$ ros2 launch moveit2_tutorials pendulum_motion_planning_pipeline_tutorial.launch.py 
[INFO] [launch]: All log files can be found below /home/jake/.ros/log/2025-01-06-08-35-19-453367-jake-Blade-46976
[INFO] [launch]: Default logging verbosity is set to INFO
/home/jake/moveit2_ws_main/install/moveit_configs_utils/share/moveit_configs_utils/default_configs/ompl_planning.yaml
/home/jake/moveit2_ws_main/install/moveit_configs_utils/share/moveit_configs_utils/default_configs/stomp_planning.yaml
[INFO] [ros2_control_node-1]: process started with pid [46980]
[INFO] [pendulum_motion_planning_pipeline_tutorial-2]: process started with pid [46982]
[INFO] [static_transform_publisher-3]: process started with pid [46984]
[INFO] [spawner-4]: process started with pid [46986]
[static_transform_publisher-3] [WARN] [1736181320.061238375] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-3] [INFO] [1736181320.110722472] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-3] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-3] from 'world' to 'base_link'
[ros2_control_node-1] [WARN] [1736181320.207723263] [controller_manager]: 'update_rate' parameter not set, using default value.
[ros2_control_node-1] [WARN] [1736181320.207908488] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-1] [INFO] [1736181320.207991888] [resource_manager]: Loading hardware 'FakeSystem' 
[ros2_control_node-1] [INFO] [1736181320.212050755] [resource_manager]: Initialize hardware 'FakeSystem' 
[ros2_control_node-1] [WARN] [1736181320.212110109] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-1] <state_interface name="velocity"> 
[ros2_control_node-1]   <param name="initial_value">0.0</param> 
[ros2_control_node-1] </state_interface>
[ros2_control_node-1] [INFO] [1736181320.212122771] [resource_manager]: Successful initialization of hardware 'FakeSystem'
[ros2_control_node-1] [INFO] [1736181320.212180380] [resource_manager]: 'configure' hardware 'FakeSystem' 
[ros2_control_node-1] [INFO] [1736181320.212189302] [resource_manager]: Successful 'configure' of hardware 'FakeSystem'
[ros2_control_node-1] [INFO] [1736181320.212194475] [resource_manager]: 'activate' hardware 'FakeSystem' 
[ros2_control_node-1] [INFO] [1736181320.212199712] [resource_manager]: Successful 'activate' of hardware 'FakeSystem'
[ros2_control_node-1] [INFO] [1736181320.228923977] [controller_manager]: update rate is 100 Hz
[ros2_control_node-1] [WARN] [1736181320.229212559] [controller_manager]: No real-time kernel detected on this system. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details on how to enable realtime scheduling.
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1736181320.285410497] [pendulum_motion_planning_pipeline]: HEREEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEEE main
[pendulum_motion_planning_pipeline_tutorial-2] 
[pendulum_motion_planning_pipeline_tutorial-2] [WARN] [1736181320.287688704] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1736181320.315256366] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.rdf_loader]: Loaded robot model in 0.0125115 seconds
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1736181320.315472557] [pendulum_motion_planning_pipeline_tutorial.moveit.core.robot_model]: Loading robot model 'jakes_pendulum'...
[pendulum_motion_planning_pipeline_tutorial-2] [INFO] [1736181320.351475284] [pendulum_motion_planning_pipeline_tutorial.moveit.ros.robot_model_loader]: Kinematics solver groups:
[... cut ...]
[pendulum_motion_planning_pipeline_tutorial-2] terminate called after throwing an instance of 'std::runtime_error'
[pendulum_motion_planning_pipeline_tutorial-2]   what():  Unable to configure planning scene monitor
[ERROR] [pendulum_motion_planning_pipeline_tutorial-2]: process has died [pid 46982, exit code -6, cmd '/home/jake/moveit2_ws_main/install/moveit2_tutorials/lib/moveit2_tutorials/pendulum_motion_planning_pipeline_tutorial --ros-args -r __node:=pendulum_motion_planning_pipeline_tutorial --params-file /tmp/launch_params_63nsdw5d'].

```

sea-bass, 2025-01-06:
Oh, nice! I had missed that while skimming.

Would this be the place to make more visible / modify?

https://moveit.picknik.ai/main/doc/how_to_guides/moveit_launch_files/moveit_launch_files_tutorial.html

Jakeisthesnake, 2025-01-06:
Yep, that would be the spot. I'll start working on it.

closed 2025-01-06 by Jakeisthesnake