moveit/moveit2_tutorials #1004: Link Constraint - Motion Planning Python API

BingPeng0, 2025-01-08:
I managed to follow the example in motion planning python API to plan and execute the robot using the joint constraints functions. But I would like to use the link constraints functions to limit my end effector orientation and move in straight line. Does anyone have any ideas to do so?

sea-bass, 2025-01-08:
It should be possible as there is also an equivalent `construct_link_constraint` function:  https://github.com/moveit/moveit2/blob/75286c7822e5f2d9babbb41b9a0ffaddb0d54808/moveit_py/moveit/core/kinematic_constraints.pyi#L5

You could try recreate the constrains in this tutorial.
https://moveit.picknik.ai/main/doc/how_to_guides/using_ompl_constrained_planning/ompl_constrained_planning.html#box-constraints

This would be a great addition to the Python tutorial if you figure this out!

BingPeng0, 2025-01-08:
I am aware of the construct_link_constraint function. 
Based on the example provided in the python motion planning example
https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html#single-pipeline-planning-custom-constraints, 
The construct_joint_constraint function is used to move the robot to specific joint values by passing the joint_values to the robot model.
So I am unclear about the purpose of the construct_link_constraint function. Is it used to specify the end-effector's pose values, or to constrain the motion to a Cartesian path or maintain a certain orientation? So, I am unsure what parameters I need to provide when using the construct_link_constraint function.
But my goal is to constraint the end-effector to move in cartesian path or certain specific orientation using Python motion planning. Any other alternatives to achieve this would be great