moveit/moveit2_tutorials #1006: motion_planning_python_api AND moveit_py error

junjunkiki, 2025-01-14:
https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html

I'm following the tutorial in this link. When I first git clone moveit2, all files should be replicated, but when I see the actual file, there's no motion_planning_python_api_tutorial. Additionally, there's no moveit_py inside packgae.xml. So I can't proceed with the api tutorial. What's the problem?
++ Additionally, after downloading the file, motion_planning_python_api_tutorial, I decompressed this file separately in src/doc/examples/. After that, I tried to launch it, but the file was missing. Can you solve the problem?
++ Moveit2 is the third git clone, and it's the same every time you do it.

[image]
[image]

sea-bass, 2025-01-14:
The `humble` branch of MoveIt2 does not have the Python API, as you discovered.

One thing you can try (though not guaranteed) is simply to comment out / remove all those offending Gazebo dependencies in the `moveit2_tutorials/package.xml` file?

I believe for Humble they are named `ign_` instead of `gz_`, but if you don't need Gazebo just remove this?

junjunkiki, 2025-01-15:
Thank you for your answer.
I git-cloned moveit2 through the humble branch instead of the main branch. Can I use the api function using Python? If it's available, I'd appreciate it if you could suggest a way

sea-bass, 2025-01-15:
The Humble branch does not have the Python API. This was released later.