moveit/moveit2_tutorials #1017: Can `MoveItPy` be used with `rclpy.node.Node`?

ymollard, 2025-02-06:
Hi,

The MoveItPy tutorials do not create a `rclpy.node.Node` instance:
https://github.com/moveit/moveit2_tutorials/blob/12edc7ea652a6590bda0fc0dc1cddfe8f0402748/doc/examples/motion_planning_python_api/scripts/motion_planning_python_api_tutorial.py#L53-L77

In my opinion this is somehow inconsistent with `rclpy` tutorials that recommand to inherit from `rclpy.node.Node` here:
https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html

That makes integration of MoveItPy with other rclpy tools (subscribers, services, ...) more complicated for beginners who cannot rely on official ROS 2 tutorials to assemble with MoveItPy.

Is there a recommanded way to use the `MoveItPy` + `rclpy.node.Node` combo?

sea-bass, 2025-02-07:
It seems like MoveItPy internally contains its own node and executor:

https://github.com/moveit/moveit2/blob/99227040ebe4bac4cc1a95afbe44ee9c543e3c02/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L55

You might be able to create something that subclasses from a node and stick the MoveItPy instance inside it as an attribute, but I haven't tried this myself.

ymollard, 2025-02-20:
Yes, it works pretty well like this: https://gitlab.com/f2m2robserv/jazzy-ros-ynov/-/blob/main/src/tiago_pick_and_place/tiago_pick_and_place/pick.py

Shouldn't the tutos create a regular node like this instead? This is confusing for newcomers, they're told in regular ROS tutorials to create a node via class inheritance, but MoveItPy is not demonstrated this way.

sea-bass, 2025-02-20:
Great to hear, and your example does look nice!

If you'd be willing to add a new tutorial on "Using MoveItPy in existing ROS 2 nodes", that would be a good addition to the currently small set of Python tutorials.