moveit/moveit2_tutorials #1033: Branch or tag for jazzy

ivanperez-keera, 2025-04-08:
This repo has branches for some ROS 2 distros.

Should there be a reference tag or branch for Jazzy?

ivanperez-keera, 2025-04-08:
Tagging @EzraBrooks for awareness.

This is currently leading to this error in the Moveit 2 CI job in the Space ROS repo:

```
#15 [stage-0  9/23] RUN cd /home/spaceros-user/moveit2/src   && sudo git clone https://github.com/ros-planning/moveit2_tutorials.git -b jazzy
#15 0.078 Cloning into 'moveit2_tutorials'...
#15 0.191 fatal: Remote branch jazzy not found in upstream origin
```

sea-bass, 2025-04-09:
There indeed is no Jazzy branch; these tutorials still run off the `main` branch here so there has been no divergence.

Since MoveIt maintainership is quite sparse these days, there have been concerns in spinning off a 3rd branch due to the additional maintenance efforts needed on top of what already isn't being met.

Right now, there is a pretty non-ideal build process that takes the Humble and Main Sphinx docs builds and smashes them together, so doing that for a 3rd branch isn't as straightforward as it could/should be. See https://github.com/moveit/moveit2_tutorials/issues/664

In short, as long as someone goes through this work and/or improves how the multi-version docs work here, that would be great! Otherwise, pointing the CI job to `main` may be easier for a quick fix.

lfurno, 2025-06-26:
Hello, I was interested in running the tutorials in jazzy because it is the ros2 distro I am planning to use. For this reason, I am modifying the humble version to adapt it to jazzy. After quite a few changes related to header files (from ".h" to ".hpp") that fixed the most of the building issues, and the addition of servo_parameters.* to moveit_servo (file taken from humble), I came across to a signature error in the example "pilts_industrial_motion_planner". It refers to the method "setPlannerId" in "moveit_task_constructor" which still has a "pipeline_name" required (I have understood that it has been removed in the "moveit2" repo).  I was wondering if it is a work that could be of any interest to you. If yes, according to your plans for  "move_task_constructor",  you would tend to modify and remove the "pipeline_name" in "move_task_constructor" too? Otherwise, I could modify the test or remove it (I have seen that it has been added with humble) or just work with the "main" version as you suggested above =D Thanks!