moveit/moveit2_tutorials #1048: GPG key error when docker build base image

from1tomax, 2025-06-09:
I want to try the demo of issac panda under path `doc/how_to_guides/isaac_panda` , but when I follow the README and try to build the docker base image , following GPG  error occurs:
```
=> ERROR [base  2/10] RUN apt-get update && apt-get upgrade -y && rosdep update                5.0s
------                                                                                               
 > [base  2/10] RUN apt-get update && apt-get upgrade -y && rosdep update:                           
0.308 Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]                      
0.438 Get:2 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]                                
0.470 Get:3 http://packages.ros.org/ros2/ubuntu jammy InRelease [4682 B]                             
0.525 Get:4 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [47.7 kB]     
0.562 Get:5 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [2984 kB]
0.571 Err:3 http://packages.ros.org/ros2/ubuntu jammy InRelease
0.571   The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
0.841 Get:6 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [4476 kB]
1.064 Get:7 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [128 kB]
1.223 Get:8 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [127 kB]
1.247 Get:9 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [1246 kB]
1.447 Get:10 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
1.547 Get:11 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
1.606 Get:12 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
1.819 Get:13 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
3.339 Get:14 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1553 kB]
3.467 Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [3295 kB]
3.762 Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [4630 kB]
4.157 Get:17 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [55.7 kB]
4.160 Get:18 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [83.2 kB]
4.167 Get:19 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [35.2 kB]
4.325 Reading package lists...
4.941 W: GPG error: http://packages.ros.org/ros2/ubuntu jammy InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
4.941 E: The repository 'http://packages.ros.org/ros2/ubuntu jammy InRelease' is not signed.
------
failed to solve: process "/bin/bash -c -o pipefail apt-get update && apt-get upgrade -y && rosdep update" did not complete successfully: exit code: 100
```

I would appreciate any helpful advice !!

Incomprehensible, 2025-12-16:
[This fix](https://github.com/ros2/ros2/issues/1696#issuecomment-2930476182) added to Docker file before `apt update` solves the issue for me.

130s, 2026-08-31:
Though not exactly the same change as https://github.com/moveit/moveit2_tutorials/issues/1048#issuecomment-3662161992 suggests, https://github.com/moveit/moveit2_tutorials/pull/1119 adds GPG update. Please see if this issue persists.

130s, 2026-09-09:
Closing as no newer issue is heard. Can be reopen if the issue still persists.

closed 2026-09-09 by 130s