moveit/moveit2_tutorials #1061: Branch humble get error with Jazzy version

kitare17, 2025-08-13:
I am a newbie learn MoveIt, I follow the document to setting project and get error
I use wsl ubuntu 24.4.01 LTS. I have the error when I try to build project below
~/ws_moveit$ colcon build --mixin release
Starting >>> moveit_resources_panda_description
Starting >>> moveit_common
Starting >>> moveit_resources_pr2_description
Starting >>> srdfdom
Starting >>> launch_param_builder
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_prbt_support
Starting >>> moveit_task_constructor_msgs
Starting >>> rosparam_shortcuts
Finished <<< moveit_resources_panda_description [0.58s]
Finished <<< moveit_common [0.58s]
Finished <<< moveit_resources_pr2_description [0.61s]
Starting >>> moveit_resources_panda_moveit_config
Starting >>> rviz_marker_tools
Finished <<< moveit_resources_fanuc_description [0.63s]
Finished <<< moveit_resources_prbt_support [0.65s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< rosparam_shortcuts [0.71s]
Finished <<< srdfdom [0.79s]
Finished <<< moveit_resources_panda_moveit_config [0.23s]
Starting >>> moveit_core
Finished <<< moveit_resources_fanuc_moveit_config [0.19s]
Starting >>> moveit_resources
Finished <<< rviz_marker_tools [0.28s]
Finished <<< moveit_task_constructor_msgs [0.99s]
Finished <<< moveit_resources [0.32s]
Finished <<< launch_param_builder [1.78s]
Starting >>> moveit_configs_utils
--- stderr: moveit_core
/home/kitare17/ws_moveit/src/moveit2/moveit_core/online_signal_smoothing/src/butterworth_filter.cpp:145:10: fatal error: pluginlib/class_list_macros.hpp: No such file or directory
  145 | #include <pluginlib/class_list_macros.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [online_signal_smoothing/CMakeFiles/moveit_butterworth_filter.dir/build.make:76: online_signal_smoothing/CMakeFiles/moveit_butterworth_filter.dir/src/butterworth_filter.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1283: online_signal_smoothing/CMakeFiles/moveit_butterworth_filter.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/home/kitare17/ws_moveit/src/moveit2/moveit_core/collision_detection/src/collision_plugin_cache.cpp:36:10: fatal error: pluginlib/class_loader.hpp: No such file or directory
   36 | #include <pluginlib/class_loader.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/build.make:188: collision_detection/CMakeFiles/moveit_collision_detection.dir/src/collision_plugin_cache.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2186: collision_detection/CMakeFiles/moveit_collision_detection.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_core [1.49s, exited with code 2]
Aborted  <<< moveit_configs_utils [0.88s]

Summary: 13 packages finished [3.00s]
  1 package failed: moveit_core
  1 package aborted: moveit_configs_utils
  1 package had stderr output: moveit_core
  40 packages not processed

Officially-Siddhant, 2025-08-14:
Having a similar issue. After looking at a recent YouTube tutorial on running MoveIt tutorials, I navigated to the comment section to find out this might be an issue with the MoveIt Core package for ROS2 Humble. Is there anymore literature on resolving this?

130s, 2026-08-18:
We decided that making `main` compatible with ROS jazzy https://github.com/moveit/moveit2_tutorials/issues/1033#issuecomment-4784020753 At the time of writing the `main` may still not build but there's a work in progress.
I'll close this as making humble branch compatible with jazzy is not in the target for now.

closed 2026-08-18 by 130s